tjko/sunspec-monitor #2: No output without the use of option Verbose mode

mcmellow, 2018-01-28:
My configuration is just a SolarEdge SE3500 HD Wave Inverter.  Using sunspec-status without the option -v result in no output.

`
/home/pi/Public$ perl sunspec-status 192.168.xxx.yyy
Time out error at /usr/local/share/perl/5.24.1/Device/Modbus/Client.pm line 82.
/home/pi/Public$ perl sunspec-status 192.168.xxx.yyy
Read of Meter 1 block registers failed (device does not support SunSpec?) at sunspec-status line 224.
/home/pi/Public$ perl sunspec-status 192.168.xxx.yyy
Read of Meter 1 block registers failed (device does not support SunSpec?) at sunspec-status line 224.
`

Using sunspec-status with the option -v result in the following output, bur with an error on Meter 1. There is no Meter installed at all.

```
/home/pi/Public$ perl sunspec-status -v 192.168.xxx.yyy
Read of Meter 1 block registers failed (device does not support SunSpec?) at sunspec-status line 180.
INVERTER:
             Model: SolarEdge  SE3500
  Firmware version: 3.2173
     Serial Number: 73000000

            Status: ON (MPPT)

 Power Output (AC):          298 W
  Power Input (DC):          303 W
        Efficiency:        98.53 %
  Total Production:      166.685 kWh
      Voltage (AC):       231.80 V (50.01 Hz)
      Current (AC):         1.40 A
      Voltage (DC):       370.10 V
      Current (DC):         0.82 A
       Temperature:        38.86 C (heatsink)
```
Sending the output to a file without the option -v results in empty file.
`perl sunspec-status 192.168.123.17  >> /tmp/SEoutput.csv
`
Sending the output to a file with the option -v results in file in text format the same as the screen output, but not a csv format.
`perl sunspec-status -v 192.168.123.17  >> /tmp/SEoutput.csv
`

tjko, 2018-01-31:
Looks like not all inverters include separate (production) meter.   I checked in change that allows specifying meter "0" to skip trying to read meter (if none installed, etc.)

Does it now work if you add option "-m 0" ?  So something like:

    sunspec-status -v -m 0 192.168.x.x

mcmellow, 2018-01-31:
With option -m 0 no errors appears. Execpt the first time time-out error. See the screen output below

```
/home/pi/Public$ perl sunspec-status -v -m 0 192.168.xxx.yyy
INVERTER:
             Model: SolarEdge  SE3500
  Firmware version: 3.2173
     Serial Number: 73000000

            Status: ON (MPPT)

 Power Output (AC):            0 W
  Power Input (DC):            0 W
        Efficiency:         0.00 %
  Total Production:      174.344 kWh
      Voltage (AC):       232.30 V (50.01 Hz)
      Current (AC):         0.00 A
      Voltage (DC):       272.30 V
      Current (DC):         0.00 A
       Temperature:        33.98 C (heatsink)

/home/pi/Public$ perl sunspec-status  -m 0 192.168.xxx.yyy
ON (MPPT),0,0,174344,232.40,0.00,371.70,0.00,34.08,0,0
```
Thanks for your support.

closed 2018-01-31 by mcmellow