tjko/sunspec-monitor #7: Error in power factor

gigantaur, 2018-08-10:
Hi,

This might be a SolarEdge/Sunspec error - apologies if so ..

I notice the numbers reported do not match the Solar edge numbers - when you have an export/import meter you get to see near eal-time values from SolarEdge which makes it easy to compare.

In essence the power factor is in error as it is showing as average of A B C  yet both B and C are at 0 amps and power factor of 1.00 (so average is not sensible)

        Real Power:         -521 W
    Apparent Power:          656 VA
      Power Factor:         0.92
      Voltage (AC):       247.83 V (50.06 Hz)
      Current (AC):         2.70 A

          'M_AC_Current' => 27,
          'M_AC_Current_A' => 27,
          'M_AC_Current_B' => 0,
          'M_AC_Current_C' => 0,
          'M_AC_Current_SF' => -1,
          'M_AC_Freq' => 5006,
          'M_AC_Freq_SF' => -2,
          'M_AC_PF' => 9209,
          'M_AC_PF_A' => 7629,
          'M_AC_PF_B' => 10000,
          'M_AC_PF_C' => 10000,

So in the example above the actual PF=.76.  I think real power should be .76 x 2.7 x 247.83 = 508.5

Apparent power also is also wrong by my calcs .. 2.7 x 247.83 = 669

cj:~ chris$  ~chris/Device-Modbus-TCP/sunspec-monitor/sunspec-status -v -d 192.168.0.125
INVERTER:
             Model: SolarEdge  SE6000
  Firmware version: 3.2305
     Serial Number: 73143E93

          'C_DeviceAddress' => 1,
          'C_Manufacturer' => 'SolarEdge ',
          'C_Model' => 'SE6000',
          'C_SerialNumber' => '73143E93',
          'C_SunSpec_DID' => 1,
          'C_SunSpec_ID' => 'SunS',
          'C_SunSpec_Length' => 65,
          'C_Version' => '0003.2305'

            Status: SLEEPING

 Power Output (AC):            0 W
  Power Input (DC):            0 W
        Efficiency:         0.00 %
  Total Production:      364.202 kWh
      Voltage (AC):       250.30 V (50.07 Hz)
      Current (AC):         0.00 A
      Voltage (DC):         0.00 V
      Current (DC):         0.00 A
       Temperature:        22.49 C (heatsink)

          'C_SunSpec_DID' => 101,
          'C_SunSpec_Length' => 50,
          'I_AC_Current' => 0,
          'I_AC_CurrentA' => 0,
          'I_AC_CurrentB' => 65535,
          'I_AC_CurrentC' => 65535,
          'I_AC_Current_SF' => -2,
          'I_AC_Energy_WH' => 364202,
          'I_AC_Energy_WH_SF' => 0,
          'I_AC_Frequency' => 50071,
          'I_AC_Frequency_SF' => -3,
          'I_AC_PF' => 10000,
          'I_AC_PF_SF' => -2,
          'I_AC_Power' => 0,
          'I_AC_Power_SF' => 0,
          'I_AC_VA' => 0,
          'I_AC_VAR' => 0,
          'I_AC_VAR_SF' => 0,
          'I_AC_VA_SF' => 0,
          'I_AC_VoltageAB' => 2503,
          'I_AC_VoltageAN' => 65535,
          'I_AC_VoltageBC' => 65535,
          'I_AC_VoltageBN' => 65535,
          'I_AC_VoltageCA' => 65535,
          'I_AC_VoltageCN' => 65535,
          'I_AC_Voltage_SF' => -1,
          'I_DC_Current' => 65535,
          'I_DC_Current_SF' => -32768,
          'I_DC_Power' => 0,
          'I_DC_Power_SF' => 0,
          'I_DC_Voltage' => 0,
          'I_DC_Voltage_SF' => -1,
          'I_Event_1' => 4294967295,
          'I_Event_1_Vendor' => 1,
          'I_Event_2' => 4294967295,
          'I_Event_2_Vendor' => 4294967295,
          'I_Event_3_Vendor' => 4294967295,
          'I_Event_4_Vendor' => 0,
          'I_Status' => 2,
          'I_Status_Vendor' => 0,
          'I_Temp_Sink' => 2249,
          'I_Temp_Sink_SF' => -2

METER (#1):
             Model: WattNode WND-3Y-400-MB
            Option: Export+Import
  Firmware version: 25
     Serial Number: 4145520

          'C_Manufacturer' => 'WattNode',
          'C_Model' => 'WND-3Y-400-MB',
          'C_Option' => 'Export+Import',
          'C_SerialNumber' => '4145520',
          'C_SunSpec_DID' => 1,
          'C_SunSpec_Length' => 65,
          'C_Version' => '25'

   Exported Energy:      161.100 kWh
   Imported Energy:      117.215 kWh
        Real Power:         -521 W
    Apparent Power:          656 VA
      Power Factor:         0.92
      Voltage (AC):       247.83 V (50.06 Hz)
      Current (AC):         2.70 A

          'C_SunSpec_DID' => 203,
          'C_SunSpec_Length' => 105,
          'M_AC_Current' => 27,
          'M_AC_Current_A' => 27,
          'M_AC_Current_B' => 0,
          'M_AC_Current_C' => 0,
          'M_AC_Current_SF' => -1,
          'M_AC_Freq' => 5006,
          'M_AC_Freq_SF' => -2,
          'M_AC_PF' => 9209,
          'M_AC_PF_A' => 7629,
          'M_AC_PF_B' => 10000,
          'M_AC_PF_C' => 10000,
          'M_AC_PF_SF' => -2,
          'M_AC_Power' => -521,
          'M_AC_Power_A' => -521,
          'M_AC_Power_B' => 0,
          'M_AC_Power_C' => 0,
          'M_AC_Power_SF' => 0,
          'M_AC_VA' => 656,
          'M_AC_VAR' => -398,
          'M_AC_VAR_A' => -398,
          'M_AC_VAR_B' => 0,
          'M_AC_VAR_C' => 0,
          'M_AC_VAR_SF' => 0,
          'M_AC_VA_A' => 656,
          'M_AC_VA_B' => 0,
          'M_AC_VA_C' => 0,
          'M_AC_VA_SF' => 0,
          'M_AC_Voltage_AB' => 24783,
          'M_AC_Voltage_AN' => 25019,
          'M_AC_Voltage_BC' => 0,
          'M_AC_Voltage_BN' => 0,
          'M_AC_Voltage_CA' => 24823,
          'M_AC_Voltage_CN' => 0,
          'M_AC_Voltage_LL' => 24803,
          'M_AC_Voltage_LN' => 25019,
          'M_AC_Voltage_SF' => -2,
          'M_Energy_VAR_SF' => -32768,
          'M_Energy_VA_SF' => -32768,
          'M_Energy_W_SF' => 0,
          'M_Events' => 0,
          'M_Exported' => 161100,
          'M_Exported_A' => 161100,
          'M_Exported_B' => 0,
          'M_Exported_C' => 0,
          'M_Exported_VA' => 0,
          'M_Exported_VA_A' => 0,
          'M_Exported_VA_B' => 0,
          'M_Exported_VA_C' => 0,
          'M_Import_VARh_Q1' => 0,
          'M_Import_VARh_Q1A' => 0,
          'M_Import_VARh_Q1B' => 0,
          'M_Import_VARh_Q1C' => 0,
          'M_Import_VARh_Q2' => 0,
          'M_Import_VARh_Q2A' => 0,
          'M_Import_VARh_Q2B' => 0,
          'M_Import_VARh_Q2C' => 0,
          'M_Import_VARh_Q3' => 0,
          'M_Import_VARh_Q3A' => 0,
          'M_Import_VARh_Q3B' => 0,
          'M_Import_VARh_Q3C' => 0,
          'M_Import_VARh_Q4' => 0,
          'M_Import_VARh_Q4A' => 0,
          'M_Import_VARh_Q4B' => 0,
          'M_Import_VARh_Q4C' => 0,
          'M_Imported' => 117215,
          'M_Imported_A' => 117215,
          'M_Imported_B' => 0,
          'M_Imported_C' => 0,
          'M_Imported_VA' => 0,
          'M_Imported_VA_A' => 0,
          'M_Imported_VA_B' => 0,
          'M_Imported_VA_C' => 0

gigantaur, 2018-08-17:
I checked - the Sunspec doc says PF is to be shown as an average .. I doubt they meant averaging unused phases though .. I suspect the simplest thing is to add a phase option to specify if single phase and/or which phase to report in

tjko, 2018-10-27:
I added --phase option (1fa0cc5b7be10bf1011beea3b633846e95e9097c) to allow selecting which phase to display the power factor for.

gigantaur, 2018-12-02:
Thanks heaps - looks good on a quick test.

(When I get time I will try to reconcile real and apparent power with SolarEdge reporting - something still not adding up)

closed 2022-04-03 by tjko