tjko/sunspec-monitor #9: Connection timeout

dpmanter, 2019-02-19:
I have an solaredge inverter SE11400. I got it to communicate over modbus TCP using this script. I was even able to write values to a database for a day. However, during my fiddling with the settings it lost connection to the SolarEdge portal. Solaredge did some magic on their end and it now connects to the solar portal. However, now I can't get a connection over modbus TCP using this script. When I execute the script it says connected on the inverter LCD but it returns a timeout error (show below)

 sunspec-status.pl -v target
Connection timed out
 at /usr/local/share/perl/5.26.1/Device/Modbus/Client.pm line 82.

Any help would be appreciated.

Thanks,
Dave

dpmanter, 2019-02-22:
More data..... if I use a modbus client application I can connect to the inverter using slave id = 2 and read the registers.

Does anyone know how to do this in the script? How do I call out slave id = 0?

lloydwatkin, 2019-02-22:
With the status comment I believe you pass meter number?

https://github.com/tjko/sunspec-monitor#syntax

dpmanter, 2019-02-22:
Thanks Lloyd but using the option -m 1 or -m 2 doesn't work.

Looking at the Modbus frame

Slave ID | Function | Start PDU Register | Length | CRC
1 byte     | 1 byte      |  2 bytes                   | 2 bytes | 2 bytes

I need to set slave id to 0x02.

dpmanter, 2019-02-22:
In the sunspec-status script if I change unit=>2 instead of unit=unit=>1

# get_inverter_common_block - return a hash containing Sunspec
#                             Common Block values
#
sub get_inverter_common_block(*) {
    my ($ctx) = @_;

    my $req = $ctx->read_holding_registers(unit=>2,
					   address=>$SUNSPEC_I_COMMON_BLOCK_START,
					   quantity=>69);
    $ctx->send_request($req) || die("send_request(): failed");

I get a partial  response from the inverter as follows:

Davids-MBP:sunspec-monitor-master manterd$ ./sunspec-status -v target
INVERTER:
             Model: SolarEdge  SE11400
  Firmware version: 3.2251
     Serial Number: 7D118B58

Connection timed out
 at /Users/manterd/perl5/lib/perl5/Device/Modbus/Client.pm line 82.

closed 2019-02-22 by dpmanter

lloydwatkin, 2019-02-22:
@dpmanter you might want to ask solaredge to look at your inverter configuration then. We had a lengthy discussion to get data from the inverter and it was all fixed by an OTA update, see https://github.com/tjko/sunspec-monitor/issues/4#issuecomment-439352510