tjko/sunspec-monitor #11: Non SunSpec Model block received

MrJoki007, 2020-01-21:
C:\Users\User\Desktop\sunspec-monitor-master>perl sunspec-status -V 192.168.178.92
INVERTER:
             Model: KACO new energy blueplanet 4.6 TL1
  Firmware version: V3.61
     Serial Number: 4.6TL10117xxxx (the x are numbers in real)

Non SunSpec Model block received (not SunSpec compliant device?) at sunspec-status line 197.

It is definetly an SunSpec compatible inverter, as seen here in the Software Downloads:
https://kaco-newenergy.com/de/produkte/blueplanet-3.0-5.0-TL1/

What im doing wrong?

tjko, 2020-01-22:
I checked in change to relax checking of the model block (8cec1ee71713480a377c9f4471fe5c18dbfd4dea), can you try it with the new version? 
Also, can you post output with the -d flag, that should show how the data returned differs from SolarEdge inverters...

MrJoki007, 2020-01-22:
`C:\Users\User\Desktop\sunspec-monitor-master\sunspec-monitor-master>perl sunspec-status -d -v 192.168.178.92
INVERTER:
             Model: KACO new energy blueplanet 4.6 TL1
  Firmware version: V3.61
     Serial Number: 4.6TL10117xxxx

          'C_DeviceAddress' => 65535,
          'C_Manufacturer' => 'KACO new energy',
          'C_Model' => 'blueplanet 4.6 TL1',
          'C_SerialNumber' => '4.6TL101178756',
          'C_SunSpec_DID' => 1,
          'C_SunSpec_ID' => 'SunS',
          'C_SunSpec_Length' => 66,
          'C_Version' => 'V3.61'

            Status:

 Power Output (AC): -1000000000000 W
  Power Input (DC): -100000000000000 W
        Efficiency:         0.00 %
  Total Production:          Inf kWh
      Voltage (AC):     65534.00 V (Inf Hz)
      Current (AC):        50.00 A
      Voltage (DC):          Inf V
      Current (DC):         0.00 A
       Temperature:         0.00 C (heatsink)

          'C_SunSpec_DID' => 0,
          'C_SunSpec_Length' => 102,
          'I_AC_Current' => 50,
          'I_AC_CurrentA' => 86,
          'I_AC_CurrentB' => 86,
          'I_AC_CurrentC' => 0,
          'I_AC_Current_SF' => 0,
          'I_AC_Energy_WH' => 4294901829,
          'I_AC_Energy_WH_SF' => 18803,
          'I_AC_Frequency' => 1,
          'I_AC_Frequency_SF' => 500,
          'I_AC_PF' => 1,
          'I_AC_PF_SF' => 1000,
          'I_AC_Power' => -1,
          'I_AC_Power_SF' => 12,
          'I_AC_VA' => -1,
          'I_AC_VAR' => 1,
          'I_AC_VAR_SF' => 0,
          'I_AC_VA_SF' => 12,
          'I_AC_VoltageAB' => 65534,
          'I_AC_VoltageAN' => 65535,
          'I_AC_VoltageBC' => 65535,
          'I_AC_VoltageBN' => 2294,
          'I_AC_VoltageCA' => 65535,
          'I_AC_VoltageCN' => 0,
          'I_AC_Voltage_SF' => 0,
          'I_DC_Current' => 0,
          'I_DC_Current_SF' => 58,
          'I_DC_Power' => -1,
          'I_DC_Power_SF' => 14,
          'I_DC_Voltage' => 65534,
          'I_DC_Voltage_SF' => 2473,
          'I_Event_1' => 262144,
          'I_Event_1_Vendor' => 65535,
          'I_Event_2' => 0,
          'I_Event_2_Vendor' => 4294967295,
          'I_Event_3_Vendor' => 4294967295,
          'I_Event_4_Vendor' => 4294967295,
          'I_Status' => 65535,
          'I_Status_Vendor' => 4,
          'I_Temp_Sink' => 93,
          'I_Temp_Sink_SF' => -32768`

So i guess its moved by 1?

MrJoki007, 2020-01-22:
Changing my $SUNSPEC_I_MODEL_BLOCK_START = 40069; to 40070 i get this which shoud be correct?

`C:\Users\User\Desktop\sunspec-monitor-master\sunspec-monitor-master>perl sunspec-status -d -v 192.168.178.92
INVERTER:
             Model: KACO new energy blueplanet 4.6 TL1
  Firmware version: V3.61
     Serial Number: 4.6TL10117xxxx

          'C_DeviceAddress' => 65535,
          'C_Manufacturer' => 'KACO new energy',
          'C_Model' => 'blueplanet 4.6 TL1',
          'C_SerialNumber' => '4.6TL101178756',
          'C_SunSpec_DID' => 1,
          'C_SunSpec_ID' => 'SunS',
          'C_SunSpec_Length' => 66,
          'C_Version' => 'V3.61'

            Status: ON (MPPT)

 Power Output (AC):          120 W
  Power Input (DC):          140 W
        Efficiency:        85.71 %
  Total Production:     4540.831 kWh
      Voltage (AC):      6553.50 V (49.90 Hz)
      Current (AC):         0.86 A
      Voltage (DC):       256.80 V
      Current (DC):         0.55 A
       Temperature:     -3276.80 C (heatsink)

          'C_SunSpec_DID' => 102,
          'C_SunSpec_Length' => 50,
          'I_AC_Current' => 86,
          'I_AC_CurrentA' => 86,
          'I_AC_CurrentB' => 0,
          'I_AC_CurrentC' => 0,
          'I_AC_Current_SF' => -2,
          'I_AC_Energy_WH' => 4540831,
          'I_AC_Energy_WH_SF' => 0,
          'I_AC_Frequency' => 499,
          'I_AC_Frequency_SF' => -1,
          'I_AC_PF' => 1000,
          'I_AC_PF_SF' => -1,
          'I_AC_Power' => 12,
          'I_AC_Power_SF' => 1,
          'I_AC_VA' => 12,
          'I_AC_VAR' => 0,
          'I_AC_VAR_SF' => 1,
          'I_AC_VA_SF' => 1,
          'I_AC_VoltageAB' => 65535,
          'I_AC_VoltageAN' => 2297,
          'I_AC_VoltageBC' => 65535,
          'I_AC_VoltageBN' => 0,
          'I_AC_VoltageCA' => 65535,
          'I_AC_VoltageCN' => 0,
          'I_AC_Voltage_SF' => -1,
          'I_DC_Current' => 55,
          'I_DC_Current_SF' => -2,
          'I_DC_Power' => 14,
          'I_DC_Power_SF' => 1,
          'I_DC_Voltage' => 2568,
          'I_DC_Voltage_SF' => -1,
          'I_Event_1' => 0,
          'I_Event_1_Vendor' => 4294967295,
          'I_Event_2' => 0,
          'I_Event_2_Vendor' => 4294967295,
          'I_Event_3_Vendor' => 4294967295,
          'I_Event_4_Vendor' => 4294967295,
          'I_Status' => 4,
          'I_Status_Vendor' => 4,
          'I_Temp_Sink' => -32768,
          'I_Temp_Sink_SF' => -1`

OleWissing, 2021-04-02:
I have the same response (Non SunSpec common block received (not SunSpec compliant device?) at ./sunspec-status line 127) when querying a SMA Sunny Island 4.4 inverter.

Maybe the same issue with starting address.